zellneralex/klipper_config #4: Q: quad_gantry_level.leveled not found

TravisWilder, 2021-05-31:
Hi

if printer.quad_gantry_level.leveled|lower == 'false'

does not work for me.
What is different on yours?
leveled seems not be an attribute of printer.quad_gantry_level

zellneralex, 2021-05-31:
Ahhh sorry that was a test of a new klipper PR https://github.com/KevinOConnor/klipper/pull/4313 and should not been merged in my repo . Sorry use the version before please. I will update it as soon the PR is merged to the latest version printer.quad_gantry_level.applied. I missed that when I added all the descriptions.

zellneralex, 2021-05-31:
If you look at the changes here https://github.com/zellneralex/klipper_config/commit/35dc4901277faca51a0a4b778f7500c7e81b3bca#diff-2b880151900936c5358548768153189701b5a73593f29078fc1f6583c17d7da6 you will find the old version

zellneralex, 2021-06-02:
It will work again If you update your klipper today and use my last commit https://github.com/zellneralex/klipper_config/commit/1959fda59038efbdcae4170d3619da5b4b34a45e

TravisWilder, 2021-06-03:
Thanks - I will test.
BTW if the QGL process aborts (like to to many retries) the PRINT_START will not complete but the print will start anyway.
Is there a way to stop the print. The print will fail as the nozzle is not even close the bed.

zellneralex, 2021-06-03:
Yes I know that’s the reason I have that macro https://github.com/zellneralex/klipper_config/blob/master/probe_qgl.cfg#L86 I Start in here https://github.com/zellneralex/klipper_config/blob/master/macro.cfg#L344 if QGL was not successfully after 5min I stop the print. That’s not optimal but the best I found as a workaround

TravisWilder, 2021-06-03:
Ok .. good to know but 5 minutes is quite long where the print already did a mess

zellneralex, 2021-06-03:
Use whatever time you like I want to make sure the QGL is in any case done.

TravisWilder, 2021-06-04:
can be closed - thanks

closed 2021-06-04 by zellneralex

TravisWilder, 2021-06-05:
Just an other idea:
Print_Start will stop if QGL exits with error.
Why not split in the slicer PRINT_START into two Commands?
FIrst PRINT_PREPARE (which will do most of what PRINT_START does now.
And PRINT_START which only will do the last steps of PRINT_START (Like PRIME LINE) but also does a check on printer.quad_gantry_level.leveled and if not true it will cancel the job.
Then the timer is not needed

zellneralex, 2021-06-05:
If you feel that is the way for you then do it. I share my approach to give ideas that is by now way the solution for anyone, that’s only the approach I am ok with