zellneralex/klipper_config #9: LCD Knop neopixel error

Lino77, 2021-11-12:
After an update, the neopixel control no longer works on the display

Error evaluating 'gcode_macro _LCD_KNOB:gcode': TypeError: coercing to Unicode: need string or buffer, Undefined found

zellneralex, 2021-11-13:
You are th3 second one … My klipper is actual and I do see no warning. I will provide you with an version tomorrow that worked for the other guy.

Lino77, 2021-11-13:
okay that's strange, got this problem on both printers. one just recently installed with mainsailos

zellneralex, 2021-11-13:
Yeah I tried different steps with the other guy … he even sends me his lcd.cfg at one point as I was unsure if he might have modified something.

zellneralex, 2021-11-13:
He wrote back to me that the modified version works for him

`
[gcode_macro _LCD_KNOB]
description: Helper: Set LCD Knob color
variable_knob: {'pri':{'R':0.0, 'G':0.3, 'B':0.0},
                'sec':{'R':0.0, 'G':0.0, 'B':0.0}}
variable_restore: {'pri':{'R':0.0, 'G':0.0, 'B':0.0},
                   'sec':{'R':0.0, 'G':0.0, 'B':0.0}}
variable_index: 2                                     ; used for blinking
variable_time: 0                                      ; timeinterval for blinking
variable_sync: 1                                      ; decide if the LED needs to be sync with movement queue
gcode:
  ##### color definition #####
  {% set color_dic =  {'OFF'  :{'R':0.0, 'G':0.0, 'B':0.0},
                       'GREEN':{'R':0.0, 'G':0.3, 'B':0.0},
                       'RED'  :{'R':0.5, 'G':0.0, 'B':0.0},
                       'BLUE' :{'R':0.0, 'G':0.0, 'B':0.5}} %}
  ##### get default parameter value #####
  {% set time = params.BLINK|default(0)|float %}
  {% set sync = params.SYNC|default(0)|int %}
  ##### get values of requested color or use default #####
  {% if 'COLOR' in params|upper %}
    {% set color = params.COLOR|upper %}
    {% set _dummy = restore.update({'pri':knob.pri}) %}
    {% set _dummy = restore.update({'sec':knob.sec}) %}
    {% if color in color_dic %}
      {% set _dummy = knob.update({'pri':color_dic[color]}) %}
    {% else %}
      {action_respond_info("LCD KNOB COLOR %s is not defined used default: GREEN" % color)}
      {% set _dummy = knob.update({'pri':color_dic.GREEN}) %}
    {% endif %}
  {% else %}
    {% set _dummy = restore.update({'pri':knob.pri}) %}
    {% set _dummy = restore.update({'sec':knob.sec}) %}
  {% endif %}
  ##### store new variable values #####
  SET_GCODE_VARIABLE MACRO=_LCD_KNOB VARIABLE=sync VALUE={sync}
  SET_GCODE_VARIABLE MACRO=_LCD_KNOB VARIABLE=time VALUE={time}
  SET_GCODE_VARIABLE MACRO=_LCD_KNOB VARIABLE=restore VALUE="{restore}"
  SET_GCODE_VARIABLE MACRO=_LCD_KNOB VARIABLE=knob VALUE="{knob}"
  ##### update to new color and start or stop blinking #####
  UPDATE_DELAYED_GCODE ID=_BLINK_DELAY DURATION={time}
  SET_LED LED=neo_display RED={knob.pri.R} GREEN={knob.pri.G}  BLUE={knob.pri.B} INDEX=2 TRANSMIT=0 SYNC={sync}
  SET_LED LED=neo_display RED={knob.pri.R} GREEN={knob.pri.G}  BLUE={knob.pri.B} INDEX=3 TRANSMIT=1 SYNC={sync}

[delayed_gcode _BLINK_DELAY]
gcode:
  ##### get variables #####
  {% set knob = printer["gcode_macro _LCD_KNOB"].knob %}
  {% set sync = printer["gcode_macro _LCD_KNOB"].sync|int %}
  {% set index = printer["gcode_macro _LCD_KNOB"].index|int %}
  {% set time =  printer["gcode_macro _LCD_KNOB"].time|float %}
  ##### end of definition #####
  {% if index == 2 %}
    SET_GCODE_VARIABLE MACRO=_LCD_KNOB VARIABLE=index VALUE=3
    SET_LED LED=neo_display RED={knob.pri.R} GREEN={knob.pri.G} BLUE={knob.pri.B} INDEX=2 TRANSMIT=0 SYNC={sync}
    SET_LED LED=neo_display RED={knob.sec.R} GREEN={knob.sec.G} BLUE={knob.sec.B} INDEX=3 TRANSMIT=1 SYNC={sync}
  {% else %}
    SET_GCODE_VARIABLE MACRO=_LCD_KNOB VARIABLE=index VALUE=2
    SET_LED LED=neo_display RED={knob.sec.R} GREEN={knob.sec.G} BLUE={knob.sec.B} INDEX=2 TRANSMIT=0 SYNC={sync}
    SET_LED LED=neo_display RED={knob.pri.R} GREEN={knob.pri.G} BLUE={knob.pri.B} INDEX=3 TRANSMIT=1 SYNC={sync}
  {% endif %}
  UPDATE_DELAYED_GCODE ID=_BLINK_DELAY DURATION={time} `

zellneralex, 2021-11-13:
Can you copy them in your cfg and try if it also solves your problem?

Lino77, 2021-11-13:
ehrlich gesagt habe ich deinen alten code genommen der funktionierte der neue aber auch

[gcode_macro _LCD_KNOB]
description: Helper: Set LCD Knob color
variable_color: 'GREEN'
variable_restore: 'GREEN'
variable_index: 2
variable_blink: 0
variable_sync: 1
gcode:
  ##### set default parameter value #####
  {% set time = params.BLINK|default(0)|float %}
  SET_GCODE_VARIABLE MACRO=_LCD_KNOB VARIABLE=sync VALUE={params.SYNC|default(1)}
  {% set var_color=printer["gcode_macro _LCD_KNOB"].color %}
  {% set var_restore=printer["gcode_macro _LCD_KNOB"].restore %}
  SET_GCODE_VARIABLE MACRO=_LCD_KNOB VARIABLE=blink VALUE={time}
  UPDATE_DELAYED_GCODE ID=_BLINK_DELAY DURATION={time}
  SET_GCODE_VARIABLE MACRO=_LCD_KNOB VARIABLE=restore VALUE='"{var_color}"'
  {% if params.COLOR == 'GREEN'%}
    SET_GCODE_VARIABLE MACRO=_LCD_KNOB VARIABLE=color VALUE='"GREEN"'
    SET_LED LED=neo_display RED=0 GREEN=0.3 BLUE=0 INDEX=2 TRANSMIT=0 SYNC={sync} 
    SET_LED LED=neo_display RED=0 GREEN=0.3 BLUE=0 INDEX=1 TRANSMIT=1 SYNC={sync}
  {% elif params.COLOR == 'RED'%}
    SET_GCODE_VARIABLE MACRO=_LCD_KNOB VARIABLE=color VALUE='"RED"'
    SET_LED LED=neo_display RED=0.5 GREEN=0 BLUE=0 INDEX=2 TRANSMIT=0 SYNC={sync}
    SET_LED LED=neo_display RED=0.5 GREEN=0 BLUE=0 INDEX=1 TRANSMIT=1 SYNC={sync}
  {% elif params.COLOR == 'BLUE'%}
    SET_GCODE_VARIABLE MACRO=_LCD_KNOB VARIABLE=color VALUE='"BLUE"'
    SET_LED LED=neo_display RED=0 GREEN=0 BLUE=0.5 INDEX=2 TRANSMIT=0 SYNC={sync}
    SET_LED LED=neo_display RED=0 GREEN=0 BLUE=0.5 INDEX=1 TRANSMIT=1 SYNC={sync}
  {% else %}
    {% if var_restore == 'GREEN'%}
      SET_GCODE_VARIABLE MACRO=_LCD_KNOB VARIABLE=color VALUE='"GREEN"'
      SET_LED LED=neo_display RED=0 GREEN=0.3 BLUE=0 INDEX=2 TRANSMIT=0 SYNC={sync}
      SET_LED LED=neo_display RED=0 GREEN=0.3 BLUE=0 INDEX=1 TRANSMIT=1 SYNC={sync}
    {% elif var_restore == 'RED'%}
      SET_GCODE_VARIABLE MACRO=_LCD_KNOB VARIABLE=color VALUE='"RED"'
      SET_LED LED=neo_display RED=0.5 GREEN=0 BLUE=0 INDEX=2 TRANSMIT=0 SYNC={sync}
      SET_LED LED=neo_display RED=0.5 GREEN=0 BLUE=0 INDEX=1 TRANSMIT=1 SYNC={sync}
    {% elif var_restore == 'BLUE'%}
      SET_GCODE_VARIABLE MACRO=_LCD_KNOB VARIABLE=color VALUE='"BLUE"'
      SET_LED LED=neo_display RED=0 GREEN=0 BLUE=0.5 INDEX=2 TRANSMIT=0 SYNC={sync}
      SET_LED LED=neo_display RED=0 GREEN=0 BLUE=0.5 INDEX=1 TRANSMIT=1 SYNC={sync}
    {% endif %}
  {% endif %}
  
[delayed_gcode _BLINK_DELAY]
gcode:
  {% set var_color=printer["gcode_macro _LCD_KNOB"].color %}
  {% set sync=printer["gcode_macro _LCD_KNOB"].sync %}
  {% if printer["gcode_macro _LCD_KNOB"].index|int == 2 %}
    SET_GCODE_VARIABLE MACRO=_LCD_KNOB VARIABLE=index VALUE=3
    {% if var_color == 'GREEN'%}
      SET_LED LED=neo_display RED=0.0 GREEN=0.3 BLUE=0.0 INDEX=2 TRANSMIT=0 SYNC={sync}
      SET_LED LED=neo_display RED=0.0 GREEN=0.0 BLUE=0.0 INDEX=1 TRANSMIT=1 SYNC={sync}
    {% elif var_color == 'RED'%}
      SET_LED LED=neo_display RED=0.5 GREEN=0.0 BLUE=0.0 INDEX=2 TRANSMIT=0 SYNC={sync}
      SET_LED LED=neo_display RED=0.0 GREEN=0.0 BLUE=0.0 INDEX=1 TRANSMIT=1 SYNC={sync}
    {% elif var_color == 'BLUE'%}
      SET_LED LED=neo_display RED=0.0 GREEN=0.0 BLUE=0.5 INDEX=2 TRANSMIT=0 SYNC={sync}
      SET_LED LED=neo_display RED=0.0 GREEN=0.0 BLUE=0.0 INDEX=1 TRANSMIT=1 SYNC={sync}
    {% endif %}
  {% else %}
    SET_GCODE_VARIABLE MACRO=_LCD_KNOB VARIABLE=index VALUE=2
    {% if var_color == 'GREEN'%}
      SET_LED LED=neo_display RED=0.0 GREEN=0.0 BLUE=0.0 INDEX=2 TRANSMIT=0 SYNC={sync}
      SET_LED LED=neo_display RED=0.0 GREEN=0.3 BLUE=0.0 INDEX=1 TRANSMIT=1 SYNC={sync}
    {% elif var_color == 'RED'%}
      SET_LED LED=neo_display RED=0.0 GREEN=0.0 BLUE=0.0 INDEX=2 TRANSMIT=0 SYNC={sync}
      SET_LED LED=neo_display RED=0.5 GREEN=0.0 BLUE=0.0 INDEX=1 TRANSMIT=1 SYNC={sync}
    {% elif var_color == 'BLUE'%}
      SET_LED LED=neo_display RED=0.0 GREEN=0.0 BLUE=0.0 INDEX=2 TRANSMIT=0 SYNC={sync}
      SET_LED LED=neo_display RED=0.0 GREEN=0.0 BLUE=0.5 INDEX=1 TRANSMIT=1 SYNC={sync}
    {% endif %}
  {% endif %}
  UPDATE_DELAYED_GCODE ID=_BLINK_DELAY DURATION={printer["gcode_macro _LCD_KNOB"].blink|float}

closed 2021-11-13 by Lino77

Lino77, 2021-11-13:
Oh, by the way, I have a BTT display, index 1 and 2 are the button, and 3 is the display

zellneralex, 2021-11-14:
Hmm I have the Fystec 1 is backlight 2,3 is the knob.there is no way to detect it