zellneralex/klipper_config #16: _CG28 and homing_override

dewi-ny-je, 2022-03-13:
Hi, thanks for your config files.

I don't understand the point in having _CG28 which homes only if needed, if homing_override already does that, on an axis by axis basis.

Is it a leftover from the past? If I remember correctly, few months I ago I didn't see the homing_override.

dewi-ny-je, 2022-03-13:
Ok I noticed now that the override is valid only for Z, but then why not using the override for all 3 axes and skip _CG28?

zellneralex, 2022-03-14:
That are 2 different thing:
- Homing overrides: Devine anything you need if you call a G28 gcode. The axis checks in there are in to insure that a G28 Z only homes Z (after a initial home of all 3 axis). 
- _CG28: Conditinal Homing: Macros that I use add the Mainsail Dashboard should run any time therefore you need an homing. But I do not want to run that homing always so there I use _CG28.

That’s the reason for both as in my opinion a G28 must be execute if requested by the user and therefor I would never add something in the homing override that would prevent a G28 or a G28 Z … request.

Best example for that is Z homing you will find that in the PRINT_START 2 times. At the start as _CG28 as that is a optional and insures only that the printer is homed once. And the second time after the nozzle clean to insure Z is homed agin with a clean nozzle. The second is done done as G28 Z and therefore is done alway.

zellneralex, 2022-03-14:
Furthermore the behavior would be different as G28 X or G28 Y do not call homing_override but G28 and G28 Z do. I did do experiments with setting all 3 axis in the override to get also a zhop when calling G28 X but the drawbacks where bigger as the gain. In normal cases I never run a single G28 X or a G28 Y.

dewi-ny-je, 2022-03-14:
I see. I got there while looking for an optimal way to ensure that homing X,Y calls a custom procedure, due to sensorless homing preferring a macro setting homing current.

I guess that, in my case, I will adapt your homing override to X and Y only, where it calls a custom routine for the axis the user explicitly requested.

I'll see if the homing override can be used to force a homing if a specific axis is requested, and operate as conditional when no axes are specified.

zellneralex, 2022-03-16:
I guess the question is answered therefore I will close it.

closed 2022-03-16 by zellneralex

dewi-ny-je, 2022-03-16:
Yes I meant to do it, sorry.

Just remembered now, I'll put it here for completeness:

> That’s the reason for both as in my opinion a G28 must be execute if requested by the user and therefor I would never add something in the homing override that would prevent a G28 or a G28 Z request.

Technically Marlin and other firmwares allow the parameter  "0" to indicate "conditional homing" equivalent to _CG28. If I implement it I'll post it here, likely in the next few days (or never...).

Thanks for the feedback and again for your macros!