zellneralex/klipper_config #19: [feature request] Filament management as separate package

dewi-ny-je, 2022-07-28:
Hi, your repo with your config is very useful and I used something here and there, but there is a feature, among the many, which I think it could be useful to separate as a fully contained package (a .conf file with all it's needed): the filament management.

As of now it relies on some parts of the main macros run at startup, and it's not that obvious which ones to extract and so on.

Maybe you could split the main macros in blocks, with comments like "following block related to filament management" or similar, so that it's easier to get what I need to put in my macros/config to get your very interesting filament management.

Of course I know that this is your configuration, provided as courtesy to us, so I don't really expect you to do it (I'm sure you are time constrained, just like I am or I would do it, even if it would take me way more time that it would to you), but at least you are aware of this feedback.
If you do it, I'll be thankful. If not, well, maybe I'll find the time one day... so far, I'm writing a macro per material, I just need to click on the one for the spool I use before starting a print. It's polluting the macro config file, but it works too.

zellneralex, 2022-07-28:
I am not sure what you mean by filament management? Do you talk about the persuade advance macros?

dewi-ny-je, 2022-07-28:
Yes, but somehow I had in my mind that you were setting temperatures too, while I see now that it's just PA.

Maybe I should then take the time (... months) and write a comprehensive management, since I would need the same for bed/nozzle temperature and extrusion multiplier (which I delegate to Klipper, not slicer software).

zellneralex, 2022-07-31:
That should only be a simple add on to the macro. You need to afdd that to the database structure and add it to the set and read macro

closed 2022-12-23 by zellneralex