zellneralex/klipper_config #21: A failed QGL in combination with MAG Probe would still run into a issue during prime line

Piranha1511, 2022-12-23:
Even with the latest commit, a failed QGL (e.g. due to a not probably docked probe) would still run into a issue during prime line and before print.

Since the CHECK_QGL will be executed after the prime line! 

Result: MAG Probe (e.g. Klicky) can smash into the bed.

zellneralex, 2022-12-23:
First off all a mag probe should not be able to smash in the bed. The G0/G1 micro should prevent that. https://github.com/zellneralex/klipper_config/blob/master/magprobe.cfg#L157.

I do not have a better solution! The add of the error code section in [virtual sdcard] may prevent that at all but I had no time yet to text it.

Piranha1511, 2022-12-28:
On my printer it is still a issue if you run a second print. A new QGL will not be performed and QGL is still seen as "true", but a bed mesh was performed before. If the probe is not probably docked after bed mesh, it will smash into the bed.

zellneralex, 2022-12-28:
I am not sure what you mean by that, but as I stated it many times. See that config as something you can get ideas.

As you see in my config I do no mesh, if you have a problem there you should work on a solution.

zellneralex, 2022-12-28:
And yes that’s the main problem with all the mag probes. If you can not get it to reliable undock you risk to smash it in the bed.
That’s the reason why my macro use an 90 degree angle to avoid that.

That’s also one reason why I will switch to Voron Tap soon.