zellneralex/klipper_config #23: Apply Z-Offset per filament

Piranha1511, 2023-01-04:
Not really a issue but more a question. You defined the Z_Adjust as parameter of Print_Start in your script. If I want define this in the Super Slicer in "Filament Settings" and not in "Printer Settings", how can I do that?

It works if I add it as parameter to PRINT_START in the "Printer Settings" but don't know how to handover the variable out of "Filament Settings". Or is there a other way to do this?

zellneralex, 2023-01-04:
That’s not possible, never thought about that use case. Would also not work as the filament gcode section is normally executed afrter the print start gcode section. So you would set a variable to a macro that is already executed.

What you could do at a 
`SET_GCODE_OFFSET Z_ADJUST=value MOVE=1 `
In your filament section, but be aware that this will be applied after the prime line

Piranha1511, 2023-01-04:
Yes, that was what I assumed. 

How do you solve this on your prints? I have for example 0.05 different offset when I switch from ASA to ABS.

zellneralex, 2023-01-04:
Hmm, I only see an difference if I use a smooth or texture plate and that is done with the plate offset.
I would add the filament type variable to start print and use a array like my plates to solve that.

goldjunge91, 2023-01-04:
> Not really an issue but more a question: You defined the Z_ADJUST as parameter for PRINT_START in your script. If I want to handover this in the Super Slicer through the "Filament Settings" and not "Printer Settings", how can I do that?
> 
> It works if I add it as parameter to PRINT_START in the "Printer Settings" but don't know how to handover the variable out of "Filament Settings". Or is there a other way to do this?

Hi Quick example 
My Start code in Cura
`START_PRINT BED_TEMP={material_bed_temperature_layer_0} EXTRUDER_TEMP={material_print_temperature_layer_0}`
this two values are from cura and will add my profile temps and send it to my start Macro
```
**{material_bed_temperature_layer_0}**
**{material_print_temperature_layer_0}**
```
```
[gcode_macro START_PRINT]
gcode:
    G21
    G90
    M82
    {% set BED_TEMP = params.BED_TEMP|default(60)|float %}
    {% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(190)|float %}
    # Start bed heating
    M140 S{BED_TEMP}
    G4 P300
    M104 S{EXTRUDER_TEMP}

```

you need to find for Klipper `{% set BED_TEMP = params.BED_TEMP|default(60)|float %}` this values  change my `BED_TEMP ` to what you need and after this 
you use your gcode for changing the Z offset and  add it like this **    M140 S. "{BED_TEMP}"** in your macro in Klipper

her are all macro and variable list from super slicer [LINK](https://github.com/supermerill/SuperSlicer/wiki/Macro-&-Variable-list )
check also this https://www.klipper3d.org/Command_Templates.html
https://klipperondocker.github.io/macros/reading_values_from_other_macros.html
I hope I have help you a little bit

Piranha1511, 2023-01-08:
Ok, thanks. Will figure it out.

closed 2023-01-08 by Piranha1511