zellneralex/klipper_config #24: Controller fan stays on after print, printer will not go into sleep

Piranha1511, 2023-01-08:
Hi Alex,

sorry for another question/issue. I really like your config and it is running now really well. Only one thing is left and you can give me maybe an advise:

My printer is not going into sleep after a print, which means my controller fan stays permanently on until I manually shutdown my printer. 

Do you have an idea where I should have a look at?

zellneralex, 2023-01-08:
Why should your printer go to sleep? Anyway look in your idle_timeout definition. Take care my config delays the stop of nevermore and the exhaust fan by 30 min to cool down the chamber and remove fume

Piranha1511, 2023-01-08:
If I print during night and the printer finished the print job, I would expect that after cool down the controller fan will also switch off. 

Yes, the nevermore and chamber is same on my printer. But still the controller fan stays on the whole night.

Piranha1511, 2023-01-08:
What I just experienced:

It works as long as I just heat up the printer and cool down again without a print. But it seems that the idle routine will not be executed if there was a print before.

zellneralex, 2023-01-08:
I do not know your config and GitHub is not the place do debug your stuff. And ask question about your config.

idle_timeout also works after printering. Maybe search for an delayed_gcode the restarts the idle timeout every time

Piranha1511, 2023-01-08:
Yes, no worries! Will figure it out.

closed 2023-01-08 by Piranha1511

zellneralex, 2023-01-08:
Try to reach me in any of the discord servers