zellneralex/klipper_config #26: Info: Special characters in JINJA2 templates can be done using HTML code

GadgetAngel, 2023-01-27:
In your README.md you have the following;

[Your README](https://github.com/zellneralex/klipper_config#unicodedecodeerror-after-update-26122021)

```
Error loading template 'gcode_macro PRINT_START:gcode': UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 2539: ordinal not in range(128)
Traceback (most recent call last): File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 51, in __init__
```

If you use Python2 or Python3 you can still use one uniform method to translate special characters.  JINJA2 template engine is meant to render HTML pages. HTML has special codes for special symbols.

So In your PRINT_START routine, you have the following:
```python
    _PRINT_AR T="{"Bed temp: act %3.1f°C min %3.1f°C (target(%3.1f°C) - delta(%2.1f°C)) %s" % 
                    (actBed,targetBed,var.temp.bed,var.delta.bed,text)}"
```

which causes errors for python2 users.  But instead of using the unicode character you could do this instead:

```HTML
    _PRINT_AR T="{"Bed temp: act %3.1f&ordm;C min %3.1f&ordm;C (target(%3.1f&ordm;C) - delta(%2.1f&ordm;C)) %s" % 
                    (actBed,targetBed,var.temp.bed,var.delta.bed,text)}"
```

the ``&ordm;`` is the special coding for HTML page for ° or the symbol called ``masculine ordinal indicator``

You can find all special characters representation for HTML at https://www.html.am/reference/html-special-characters.cfm

Here is another example, you have this in your CANCEL_PRINT:

```python
      {action_respond_info("Extruder Temp to low heat to %3.1f°C" % printer.configfile.settings.extruder.min_extrude_temp)}
```

But it can be written as follows and will work no matter if you are using python2:

```HTML
      {action_respond_info("Extruder Temp to low heat to %3.1f&ordmC" % printer.configfile.settings.extruder.min_extrude_temp)}
```
Here is the HTML code for the degree symbol 

```HTML
° = &ordm
```
Below is my output to the UI console:
[image]

Here is the code I used:

```
M118 Chamber Temp set to {S}&ordm;C
```
I hope this helps other.

zellneralex, 2023-01-27:
Thank you for that. I will add it to the readme. 

Honestly Python2 installation need to “die”. Kiauh now supports setting up a Python3 based Klipper so you can switch in Minutes. And MainsailOS also setups a Python3.

GadgetAngel, 2023-01-27:
> Thank you for that. I will add it to the readme.
> 
> Honestly Python2 installation need to “die”. Kiauh now supports setting up a Python3 based Klipper so you can switch in Minutes. And MainsailOS also setups a Python3.

I used that feature in KIAUH  (for python3), but if you look at /home/pi/klippy-env/bin/ path which is the Virtual Environment for Klipper I still see the python2 is still installed.

I do not know if all the Klipper .py files are converted over to python3.  I thinks some are not yet done yet.

Do you know?

zellneralex, 2023-01-27:
We checked kiauh it should build a new env based on py3. But I will talk to the kiauh def and test it with him again

No there are still a few scripts that make problems but that are debug scripts that the regular user does not use at all.

GadgetAngel, 2023-01-27:
> We checked kiauh it should build a new env based on py3. But I will talk to the kiauh def and test it with him again. Can you describe how you did the switch between 2 and 3.
> 
> No there are still a few scripts that make problems but that are debug scripts that the regular user does not use at all.

Let me see if I can find it.

GadgetAngel, 2023-01-27:
> We checked kiauh it should build a new env based on py3. But I will talk to the kiauh def and test it with him again. Can you describe how you did the switch between 2 and 3.
> 
> No there are still a few scripts that make problems but that are debug scripts that the regular user does not use at all.

Well, the doc in no longer there.  

What I recall is that they used KIUAH can used the option to use a custom Klipper branch that was updated to Python3.

I suppose Bigtreetech took it down because, that was over a year ago and the branch probably did not stay up to date with Klipper.

From what I  have found out today is that Klipper mainly is still pyton2 but is moving to python3.  So Klipper does use both versions of python .  The last entry I saw was that Kevin was saying that python3 was slower with some things in Klipper than python2 and that is why the move has not been made yet.

If you know differently, please inform me.

zellneralex, 2023-01-28:
Yes Kevin’s tells more than a year he is switching to py3. The main problem is that a Debian 11 (bullseye) some packages are not available anymore. That’s e.g. the reason why the switch for the IS post processing script to py3 was mandatory.

The process to get a machine to py3 using kiauh is very simple:
- update kiauh 
- Open it
- Remove Klipper
- Install Klipper and select the Python3 virtual environment in that process

Doing that you will see that your v-env only contains Python3. That has nothing to do with that your current of has installed both. But if you would do a fresh install with a bullseye lite image you would see that Python 2 is not installed at all.

zellneralex, 2023-01-28:
I just checked we have more than 10k MainsailOS 1.0.1 and ~7k MainsailOS 1.0.0 downloads. So at least all these machines run in a Python3 only environment.

zellneralex, 2023-01-28:
And much worse for Python 2.7 😉 https://devguide.python.org/versions/#python-release-cycle.That was EOL 2020.