zellneralex/klipper_config #28: Filament runout sensor

Orob-Maslow, 2024-01-25:
Tried your filament sensor code and klipper chokes on:

{% set _dummy = filament_sensor.update({params.SENSOR|string: params.ENABLE|int}) %}

stating that there is no .update in jinja.  Any thoughts on that one?

zellneralex, 2024-01-25:
That code is working with mainline Klipper in a python3 env. So I am not sure why it should not work for you.

Orob-Maslow, 2024-01-25:
wow, thanks for the immediate response.  I'm running mainline klipper from kiauh install on RPI3B+ with an BTT octopus board and a BTT v2 smart filament sensor and your code.  Klipper might be one update behind right now (v11, vs v12), but the last update was over the weekend.  The sfs shows up on the screen as enabled, but klipperscreen kicks this .update jinja error every restart.  Wondering if it is just my setup or if you've seen this before.  Seems to work anyway though, so I commented that line out.  I don't really undestand what the _dummy line does anyway and that is the only reference to it in the entire printer.cfg.

zellneralex, 2024-01-25:
Never seen that before. .update is a base jira function.

ok 
`
{% set _dummy = filament_sensor.update({params.SENSOR|string: params.ENABLE|int}) %}
`

The _dummy is a workaround here. Klipper needs a left side argument. There would be a do extension for jira that Klipper does not want to add, therefor we need the workaround. The important part is

`
filament_sensor.update({params.SENSOR|string: params.ENABLE|int})
`
That modifies the existing entry of the already existing sensor dictionary entry.

Attach a fresh klippy log if you use Mainsail you can use the rollover feature in the setting app to get a fresh log. Then I will take a look if I see something

zellneralex, 2024-01-25:
So that is in my actual running printer
[image]
I only shorten _dummy to have it equal in all configs

zellneralex, 2024-01-25:
https://github.com/Klipper3d/klipper/pull/5149 That’s the old Klipper PR that explains a little better why we using this wotrkaround

Orob-Maslow, 2024-01-26:
[Klippy.log.txt](https://github.com/zellneralex/klipper_config/files/14059831/Klippy.log.txt)

the _Dummy isn't the issue.  it is the .update method that is apparently the problem.  In the PR for klipper main you link to, the _dummy line uses a .append method, not a .update, so the _dummy blocks the console message, but it isn't the same problem exactly I don't think.

[image]